EiNSTeiN-/roadrunner-filament-sensor #1: Communication between the sensor/host via USB

mainzerp, 2024-01-28:
Hi,

would i be possible to add a firmware variant to let the sensor communicate via usb?

-Schokoladenbrezel

EiNSTeiN-, 2024-01-28:
Hi! Thanks for the suggestion, I've implemented the option with https://github.com/EiNSTeiN-/roadrunner-filament-sensor/commit/0db8af00ef29fb4923cfbe672934222d15fd7dec and https://github.com/EiNSTeiN-/roadrunner-filament-sensor/commit/b78dd77db461f2e9db71afce3074a9d35e08fd71 but this new code is still untested. I'll upload the compiled firmware as soon as I can find some time to validate the new functionality.

mainzerp, 2024-01-29:
Thanks for the fast reply. If you need someone for testing, let me know.

SartorialGrunt0, 2024-01-31:
I just ordered hardware and would be willing to test as well. USB would be easier to route then I2C.

EiNSTeiN-, 2024-02-01:
I added a new commit b7330ef7bf16621f5a04c85b01e746bac25c9b6f, tested reading data with the sensor successfully and added the firmware to the 1.0 release here: https://github.com/EiNSTeiN-/roadrunner-filament-sensor/releases/tag/v1.0

EiNSTeiN-, 2024-02-01:
Thanks again for the suggestion, I'll close this one as done but let me know if there is an issue. I likely won't be running my own sensors over serial so I'm unlikely to find bugs in this part of the code.

closed 2024-02-01 by EiNSTeiN-