EiNSTeiN-/roadrunner-filament-sensor #19: 'PrinterExtruder' object has no attribute 'move'

jkwilling, 2025-08-24:
https://github.com/Klipper3d/klipper/commits/master/klippy/kinematics/extruder.py

Traceback (most recent call last):
  File "/home/jwilling/klipper/klippy/klippy.py", line 164, in _connect                                                                         cb()
  File "/home/jwilling/klipper/klippy/extras/high_resolution_filament_sensor.py", line 812, in _handle_ready
    self.orig_extruder_move = self.extruder.move
                              ^^^^^^^^^^^^^^^^^^
AttributeError: 'PrinterExtruder' object has no attribute 'move'
Transition to shutdown state: Internal error during ready callback: 'PrinterExtruder' object has no attribute 'move'
---------------------------------------------------------------------------------------
Klipper was updated on May 12..
   [extruder: Rename extruder.move() to extruder.process_move()] (https://github.com/Klipper3d/klipper/commit/f06eeb5c7aa5b297d0b6128b8cd878a35b6ea89d)
[KevinOConnor](https://github.com/Klipper3d/klipper/commits?author=KevinOConnor)
KevinOConnor committed on May 12

There was also a change in arguments  on the same day:
[toolhead: Initial support for adding extra axes to toolhead moves](https://github.com/Klipper3d/klipper/commit/bb281834b07cc5bc8c892e59bb701a9a079009b0)
[KevinOConnor](https://github.com/Klipper3d/klipper/commits?author=KevinOConnor)
KevinOConnor committed on May 12

---------------------------------------------------------
I updated line 812 to be:
self.orig_extruder_move = self.extruder.process_move

Also, to add the additional argument, I edited line 723 & 727 to have ea_index at the end:
723     def _capture_extruder_move(self, move_time, move, ea_index):
727         return self.orig_extruder_move(move_time, move, ea_index)

CookiezRGood, 2025-08-24:
I made a pull request with this like 3 months ago, and it hasn't been merged yet. I'm not sure if @EiNSTeiN- knows about it or not, but I hope they merge it eventually. For now, my printer is just referencing my forked code so that it doesn't have error messages in my updates

EiNSTeiN-, 2025-08-24:
Should be solved with #16

closed 2025-08-24 by EiNSTeiN-